xboxdrv/xboxdrv #112: Steam won't detect controller?

P1an3t, 2015-01-03:
I have a wired xbox 360 controller and am running ubuntu 14.04. I have xboxdrv installed and when i run "sudo xboxdrv --detach-kernel-driver" everything seems to work. I also have the jstest-gtk thing installed and it shows everything on my controller working correctly. Then when I launch steam and play a game the controller doesn't work. Also when I go into big picture mode settings it says no controller detected. Any idea what I can do?
Thanks,
P1an3t

jrasky, 2015-02-01:
My experience is you need to specifically rmmod xpad, and run xboxdrv with --mimic-xpad

Grumbel, 2015-11-06:
Either use `--mimic-xpad` or set:

    export SDL_GAMECONTROLLERCONFIG="0000000058626f782047616d65706100,Xbox Gamepad (userspace driver),platform:Linux,a:b0,b:b1,x:b2,y:b3,start:b7,back:b6,guide:b8,dpup:h0.1,dpdown:h0.4,dpleft:h0.8,dpright:h0.2,leftshoulder:b4,rightshoulder:b5,lefttrigger:a5,righttrigger:a4,leftstick:b9,rightstick:b10,leftx:a0,lefty:a1,rightx:a2,righty:a3"

This should cause most games to recognize the controller without the use of `--mimic-xpad`.

closed 2015-11-06 by Grumbel