xboxdrv/xboxdrv #116: Map an axis on to two separate axes?

YungDaVinci, 2015-01-25:
I've been trying to get my dancepad to work correctly, because the buttons are acting as axes. I tried mapping them to the dpad_x and dpad_y axes, then using the --dpad-as-button to get them to work like buttons. While they do show up as buttons, I can't press left and right or up and down at the same time, as if they are still an axis.
Is there any way to do this? If not, could it be implemented? Thanks.

Grumbel, 2015-01-25:
What kind of dance mat is that? Does that show up as Xbox360 controller or as regular USB controller/adapter that you are trying to remap with `--evdev`?

In the regular Xbox360 protocol the dpad is just a bunch of bits for up/down/left/right, so using them as separate buttons should work just fine (unless there is a bug somewhere).

Some other joysticks/adapters however report the dpad as 8 directions and a center position, in those cases it wouldn't be possible to handle up/down as completely separate buttons

YungDaVinci, 2015-02-01:
I think my mat is doing the 8 directions thing. The dance mat shows up as a usb controller.