xboxdrv/xboxdrv #118: xboxdrv have wrong mapping with xbox360 wireless controller

lucquera, 2015-02-23:
I'm using Fedora 21 (32-bit), and some Steam games. I have a Xbox 360 wireless controller, which works fine with the xpad module, except for the led lights. I've installed the xboxdrv 0.8.5, and the leds works fine, but the controller is all messed up, the button mapping all wrong and weird. I've tested with Super Meat Boy, Bastion and Fez. In SMB, the controller is not even readed (it worked with xpad module), in Bastion, the LT and RT are analogs, B is A, A is B, and is impossible to play due to LT behavior. In Fez it was just inverted: B=A, A=B, X=Y, Y=X, RT=RB, LT=LB.
I don't know why this is happening, cause is not a generic usb joystick, but a official Xbox 360 wifi controller. I must remap all the buttons?
p.s: using evtest, the Dpad is readen like digital buttons, and LT/RT are readen like analog ABS_Z.

Grumbel, 2015-02-23:
Use `xboxdrv --mimic-xpad` or `xboxdrv --mimic-xpad-wireless` to make it behave similar to the xpad driver, that should make it work with most games work.

Going into Steam's Big Picture Mode and setting up the controller there should also make it work with most games, as Steam supplies the controller configuration from Big Picture Mode to the game (requires that the game uses the SDL2 GameController API).

closed 2015-02-23 by Grumbel

lucquera, 2015-02-28:
That fixed my problem. But now, if i run this command alongside the daemon option (xboxdrv -D --silent --mimic-xpad), some games don't recognize the controller, and everything get messed up again.