xboxdrv/xboxdrv #119: xboxdrv is not taking js0

adolson, 2015-03-07:
This is a bit odd... I have a udev rule set up to block js* devices from being created by my ASUS Gamepad. When I run xboxdrv, it tells me that the virtual device is available at js0, but it's not - it's actually js1. I set up a script to mv js1 to js0, but this is a bug, is it not?

Grumbel, 2015-03-08:
What js device xboxdrv displays is just an educated guess, it doesn't actually know where it really ends up, as uinput doesn't provide that information. There might be a way to figure that out via the /sys/  sysfs filesystem, but I haven't really looking into that.

adolson, 2015-03-08:
OK, that makes sense. But any idea why it isn't picking the first-available js* device?

beaumanvienna, 2015-03-08:
I can confirm this behavior, it may happen on some hardware. One example is actually my Acer laptop. In our scripts, we were previously assuming that the first controller is mapped to js0, but we had to change it.  As a work-around, we now  search for the  lowest js device number. Starting from that index the numbering is sequential for all connected game controllers.

adolson, 2015-04-26:
I've been working again on trying to get my system to handle multiple controllers that require xboxdrv, so I've done a lot more reading again. In the Arch wiki, I found that even if your udev rules are set up to rm /dev/input/js0, the device number will still be reserved. So this is likely why xboxdrv is taking js1.

I can't find a way to work around this problem. I'm still trying, but it's a huge headache.

I have written some scripts and been messing with it, so I had it at the point where powering on two ASUS Gamepads will delete the native js* devices, rename the event* devices to nexus* (to prevent problems with Steam but still allow xboxdrv to act on them), then launch xboxdrv with a config file and passing in the nexus* device... And finally, rename the js* device to the lowest currently-available js* device. THIS WORKS... Until one gamepad loses power, then all predictability goes out the window because - and I'm guessing here - udev seems like it's still trying to delete the device's original js* device (which was already deleted) which may actually be another pad's xboxdrv js* device...

I can't figure out how to make udev PROPERLY rename a device (seems like this was an option at one point, and then taken away), nor can I figure out how to make it ignore a device and not create a js* file at all (seems like this was also an option at one point, and also taken away...)...

So my next plan of attack is to delete the native js* devices for the ASUS Gamepads (js0, js2, for example), and rename the js* devices created by xboxdrv (js1, js3, for example) to joystick* with the same number, and then create js* symlinks to those (js0 -> joystick1, js1 -> joystick3) both in the add AND remove udev events, to work around the bugginess I mentioned earlier.

If this all sounds convoluted, it is. If it sounds complicated, it is. If it sounds like it could be done easier, I hope you're right and would LOVE to know how.

beaumanvienna, 2015-04-26:
Why not live with it? This is just handle after all. When we faced this problem half a year ago, we came to the conclusion that we have two possibilities. Debug the kernel or [program around it](https://github.com/ProfessorKaos64/RetroRig/blob/master/scriptmodules/gamepads.shinc#L393). We chose the low hanging fruit.

adolson, 2015-04-26:
I don't have source code access to the Steam games that I want to play, so "program around it" is not a solution, as far as I can tell.

beaumanvienna, 2015-04-26:
Have a look at [antimicro](https://github.com/Ryochan7/antimicro), that's how we made it work in RetroRig.

adolson, 2015-04-26:
I don't see how that is going to solve this... Games I am playing on what
is essentially an equivalent to a Steam Machine have gamepad support built
in, just some of them look for js0. Emulating a keyboard and mouse is going
to cause a whole lot more issues that I don't want to deal with, not the
least of which is having to use a keyboard and mouse to set up profiles for
every game.

This is a problem with udev, and it will be solved with udev. I am 98% of
the way there already, and I'm not giving up now.

gyates100895, 2015-04-26:
@adolson my fork of Ubuntu_xboxdrv handles this pretty well. I believe that it starts js naming after whatever system ones are present but it works well from there. If you figure it out though please share :) maybe contribute to systemd_xboxdrv when this is all finished, as I am currently facing a similar issue.

techwiz24, 2015-05-07:
@adolson I'm experiencing this problem as well with the ASUS Game Pad. If you ever get it working, I'd love to see what you did (I'm also on arch)

adolson, 2015-05-07:
I did get it working. You probably got a notification from GitHub a few minutes ago from your post on my post on the Valve tracker which sums up basically what I had to do. I will upload my scripts to my own GitHub account in a few minutes so you can see how I do it. I'm on Debian, but should be easy for you to adapt to Arch.