xboxdrv/xboxdrv #120: joystick motion causes program to crash?

kevinfol, 2015-03-13:
I am trying to set up a non-xbox 360 controller look like an xbox controller using the '--mimic-xpad' option and doing the button and axis mapping manually. However, when I run xboxdrv with the command:

 sudo xboxdrv --evdev /dev/input/event0 --evdev-absmap ABS_RX=X2,ABS_RY=Y2,ABS_HAT0X=X1,ABS_HAT0Y=Y1,ABS_X=DPAD_X,ABS_Y=DPAD_Y --evdev-keymap BTN_X=A,BTN_B=B,BTN_Y=X,BTN_A=Y,BTN_TR2=RB,BTN_TR=RT,BTN_TL2=LB,BTN_TL=LT,BTN_START=Start,BTN_SELECT=Back,BTN_THUMBR=TL,BTN_THUMBL=TR --mimic-xpad

the program responds correctly until I move any one of the axes (inluding the dpad) to its maximum position. at which point the program crashes with the error : 
"xboxdrv: src/helper.cpp:158: float to_float(int, int, int): Assertion `value >= min' failed."

Is there any way to resolve this issue? Thanks

kuar-z, 2015-03-30:
Same issue when using evdev for input mapping on a PS# connected via bluetooth. Only seems to happen when hitting -32768 on an x or y axis value, positive values don't seem to crash, at least with my setup. I've tried setting max numbers with calibration to no effect as well adjusting sensitivity. If the axis is not mapped from the input then a crash does not occur.

Here is the command I'm working off of:
````
--evdev /dev/input/event1 
--evdev-keymap KEY_#300=y,KEY_#302=a,KEY_#301=b,BTN_DEAD=x,BTN_TOP=start,BTN_TRIGGER=back,BTN_A=guide,BTN_BASE6=rb,BTN_BASE5=lb,BTN_TOP2=du,BTN_BASE2=dl,BTN_BASE=dd,BTN_PINKIE=dr,BTN_THUMB=tl,BTN_THUMB2=tr 
--evdev-absmap ABS_#13=rt,ABS_#12=lt,ABS_X=x1,ABS_Y=y1,ABS_Z=x2,ABS_RX=y2 
--axismap -Y1=Y1,-Y2=Y2 
--calibration RT=0:127:255,LT=0:127:255 
--mimic-xpad-wireless

abuisine, 2015-10-02:
Same thing here (raspbian on raspberry pi 2) with this configuration :
[xboxdrv]
#ui-clear = true
evdev = /dev/input/event2
extra-devices = false
extra-events = false
deadzone = 6000
device-name = "Microsoft Xbox 360 Wireless Controller"

[evdev-absmap]
ABS_X=x1
ABS_Y=y1
ABS_Z=x2
ABS_RZ=y2
ABS_#48=lt
ABS_#49=rt

[evdev-keymap]
BTN_THUMB=tl
BTN_THUMB2=tr
BTN_BASE5=lt
BTN_BASE6=rt
KEY_#302=b
KEY_#301=a
BTN_DEAD=y
KEY_#300=x
BTN_TRIGGER=back
KEY_#720=guide
BTN_TOP=start
BTN_TOP2=du
BTN_PINKIE=dr
BTN_BASE=dd
BTN_BASE2=dl

NonlinearFruit, 2017-07-30:
I have the same problem. I run it with a config via `sudo xboxdrv -c example.xboxdrv --silent`. And it seems to work fine as long as I don't use the joysticks. But when I use a joystick (for `wasd` or for moving the mouse), xboxdrv freezes. Is there anyway to fix this?

theonedemon, 2018-05-04:
i have some problem with game "Hollow Knight"
run xboxdrv:
`./xboxdrv --evdev /dev/input/event13 --detach-kernel-driver --mimic-xpad-wireless --calibration RT=0:127:255,LT=0:127:255`

on start game after 4 sec game crash

commit 605e45ed (2015-11-09): Don't assert() in to_float(), just clamp the return value

Fixes #120 might still be worth to investigate why the values go out of range in the first place
commit 08dcf4c2 (2015-11-09): Don't assert() in to_float(), just clamp the return value

Fixes #120 might still be worth to investigate why the values go out of range in the first place