xboxdrv/xboxdrv #121: Cannot set force-feedback in the config file

RalfJung, 2015-03-31:
I am trying to write a config file corresponding to the following manual CLI call: `sudo xboxdrv  --trigger-as-zaxis --axismap "-z=z" --force-feedback`. My current attempt is:
```
[xboxdrv]
trigger-as-zaxis = true
force-feedback = true
[axismap]
-Z=Z
```

xboxdrv reacts to this config file by saying:
```
-- [ ERROR ] ------------------------------------------------------
bad lexical cast: source type value could not be interpreted as target
```

Following the xample file, I also tried `force-feedback = ` (without giving it a value), to no avail. It seems not to be possible to configure force-feedback in the config file?

Grumbel, 2015-04-01:
force-feedback = 1

Seems to work. Looks like boost::lexical_cast<bool> isn't taking a `true`:

options.cpp:338:  get_controller_slot().set_force_feedback(boost::lexical_cast<bool>(value));

The error message could also need some improvements.

RalfJung, 2015-04-02:
>     force-feedback = 1
> 
> Seems to work.

Indeed it does, thanks! Strange enough, "true" works fine for
trigger-as-zaxis...

Grumbel, 2015-04-02:
Not so strange if you look at the inner workings of code ;-)

`force-feedback` is a good old bool variable and (misbehaves) as expected.

`zaxis-as-trigger=true` on the other side is really a function call in disguise and just a shortcut for `--ui-axismap trigger=ABS_Z:-255:255:0:0,lt=void,rt=void`. The function handling takes a different path then the variable handle and doesn't have that bug.

This also means that something like `zaxis-as-trigger=false` is pointless, it simply doesn't do anything and doesn't undo the effect of an earlier `zaxis-as-trigger=true`. A `force-feedback=0` on the other side would work as expected. It's a weird artefact of the `.ini` style syntax, but necessary to preserve the order of instructions.

RalfJung, 2015-04-03:
> Not so strange if you look at the inner workings of code ;-)

As usually ;-) . Thanks for the explanation, and more importantly thanks
for this great tool!