xboxdrv/xboxdrv #133: Wireless Chatpad Support

Dreadlow, 2015-06-28:
I have a quick question in regards to Chatpad support.

I noticed the toggle for --chatpad to enable the pad on the manpage. I thoroughly looked and it didn't mention anything in particular about the wireless controller. I know that there hasn't been much drive to get the chatpad to work with a wireless controller (based on google searches and general research), however I was curious to know if this feature should be working for wireless controllers and not just USB.

I recently was able to setup moonlight embedded on a Raspberry Pi while using xboxdrv to use my wireless controller, and after finding a cheap chatpad at Gamestop I was curious to see if I could get it to work for use with Steam's Big Picture mode and chat. It's more than likely a niche situation, but chatpad and headset support is something that could really change how Steam gets accepted into the living room.

Is the lack of wireless chatpad support because it's not implemented in xboxdrv, did I perhaps miss something in a configuration or toggle and it should be working, and/or is this development still in progress, put on hold, etc?

Thanks.

Grumbel, 2015-11-03:
[Copy&paste from an old mail]

There is currently no active development going on with xboxdrv and
there hasn't been for two or three years, so don't expect chatpat
support anytime soon. That said, I haven't given up on, just haven't
found the motivation to do so as I don't have much need for it at the
moment. I don't think there where any real show stopper for support of
the wireless chatpad, the info in [1] should make it pretty easy to
have the chatpad react and do something. The problem was more that
xboxdrv couldn't really deal with all the additional buttons of the
chatpad and the larger rewrite to make xboxdrv more flexible in that
regard kind of stalled before it got finished, that's why there is
currently the messy situation with a 'stable' branch in git and a half
working 'develop' branch. But the chatpad protocol should be mostly
figured out and the info in [1] should allow you to make the LED light
up.

Some links with info (some of which you probably already know):

[1] https://code.google.com/p/chatpad-super-driver/issues/detail?id=46&q=wireless
[2] https://code.google.com/p/xbox360wirelesschatpad/source/checkout
[3] https://github.com/d235j/360Controller/tree/master/360Controller
[4] http://www.mp3car.com/hardware-development/140189-contest-xbox-chatpad-driver-challenge-44.html

darkbibble, 2015-11-17:
having the chatpad working with the wireless pad would be great for steam OS use.
at the moment it uses on-screen keyboard with the wireless gamepad, but I think the chatpad would extremely usefull in steam OS

charlesmillet, 2015-12-17:
chatpad on wireless xbox controller would be very useful on a lot of thin especially Steam and HTPC. Thank you for your work.

dimka665, 2016-11-12:
I need this feature too.