xboxdrv/xboxdrv #140: Can't compile (not stable not devel versions)

alejandro-anv, 2015-08-27:
I am trying to build a packet for mageia 5 <b>64 bits</b>. Tried to compile the source that was previously working but I receive the following error:

libx11-dev must be installed!

But libx11-devel <b>is</b> installed (I double checked it, the version is lib64x11-devel-1.6.2-5 ). I commented the line in SConstruct. Then compilation starts OK but aborts with this error:
<pre>
scons: Building targets ...
g++ -o src/xboxdrv_g_controller.o -c -isystem/usr/lib64/dbus-1.0/include -isystem/usr/include/dbus-1.0 -isystem/usr/include/glib-2.0 -isystem/usr/lib/glib-2.0/include -pthread -isystem/usr/include/libusb-1.0 -g -O3 -Wall -ansi -pedantic -DPACKAGE_VERSION='"0.8.5"' -Isrc src/xboxdrv_g_controller.cpp
In file included from /usr/lib/glib-2.0/include/glibconfig.h:9:0,
                 from /usr/include/glib-2.0/glib/gtypes.h:32,
                 from /usr/include/glib-2.0/glib/galloca.h:32,
                 from /usr/include/glib-2.0/glib.h:30,
                 from /usr/include/glib-2.0/gobject/gbinding.h:28,
                 from /usr/include/glib-2.0/glib-object.h:23,
                 from src/xboxdrv_g_controller.hpp:22,
                 from src/xboxdrv_g_controller.cpp:19:
src/xboxdrv_g_controller.cpp: En la función ‘GType xboxdrv_g_controller_get_type()’:
src/xboxdrv_g_controller.cpp:36:1: error: el tamaño de la matriz ‘_GStaticAssertCompileTimeAssertion_0’ es negativo
 G_DEFINE_TYPE(XboxdrvGController, xboxdrv_g_controller, G_TYPE_OBJECT)
 ^
src/xboxdrv_g_controller.cpp:36:1: error: el tamaño de la matriz ‘_GStaticAssertCompileTimeAssertion_1’ es negativo
 G_DEFINE_TYPE(XboxdrvGController, xboxdrv_g_controller, G_TYPE_OBJECT)
 ^
src/xboxdrv_g_controller.cpp:36:1: error: el tamaño de la matriz ‘_GStaticAssertCompileTimeAssertion_2’ es negativo
 G_DEFINE_TYPE(XboxdrvGController, xboxdrv_g_controller, G_TYPE_OBJECT)
 ^
scons: *** [src/xboxdrv_g_controller.o] Error 1
scons: building terminated because of errors.
</pre>

Then, I downloaded the development version and I receive the same error when compiling:
<pre>
[  0%] Building CXX object CMakeFiles/libxboxdrv.dir/src/xboxdrv_g_daemon.cpp.o
In file included from /usr/lib/glib-2.0/include/glibconfig.h:9:0,
                 from /usr/include/glib-2.0/glib/gtypes.h:32,
                 from /usr/include/glib-2.0/glib/galloca.h:32,
                 from /usr/include/glib-2.0/glib.h:30,
                 from /usr/include/glib-2.0/gobject/gbinding.h:28,
                 from /usr/include/glib-2.0/glib-object.h:23,
                 from /tmp/xboxdrv-develop/src/xboxdrv_g_daemon.hpp:22,
                 from /tmp/xboxdrv-develop/src/xboxdrv_g_daemon.cpp:19:
/tmp/xboxdrv-develop/src/xboxdrv_g_daemon.cpp: In function 'GType xboxdrv_g_daemon_get_type()':
/tmp/xboxdrv-develop/src/xboxdrv_g_daemon.cpp:25:1: error: size of array '_GStaticAssertCompileTimeAssertion_0' is negative
 G_DEFINE_TYPE(XboxdrvGDaemon, xboxdrv_g_daemon, G_TYPE_OBJECT)
 ^
/tmp/xboxdrv-develop/src/xboxdrv_g_daemon.cpp:25:1: error: size of array '_GStaticAssertCompileTimeAssertion_1' is negative
 G_DEFINE_TYPE(XboxdrvGDaemon, xboxdrv_g_daemon, G_TYPE_OBJECT)
 ^
/tmp/xboxdrv-develop/src/xboxdrv_g_daemon.cpp:25:1: error: size of array '_GStaticAssertCompileTimeAssertion_2' is negative
 G_DEFINE_TYPE(XboxdrvGDaemon, xboxdrv_g_daemon, G_TYPE_OBJECT)
 ^
CMakeFiles/libxboxdrv.dir/build.make:139: recipe for target 'CMakeFiles/libxboxdrv.dir/src/xboxdrv_g_daemon.cpp.o' failed
make[2]: *** [CMakeFiles/libxboxdrv.dir/src/xboxdrv_g_daemon.cpp.o] Error 1
CMakeFiles/Makefile2:60: recipe for target 'CMakeFiles/libxboxdrv.dir/all' failed
make[1]: *** [CMakeFiles/libxboxdrv.dir/all] Error 2
Makefile:117: recipe for target 'all' failed
make: *** [all] Error 2
</pre>

Any help please?

Grumbel, 2015-08-27:
Something wrong with your glib install maybe, mixing 32bit header and 64bit library or so. See (and other stuff found on Google):

https://mail.gnome.org/archives/gtk-app-devel-list/2012-October/msg00052.html

alejandro-anv, 2015-08-28:
Yo are right, I had wrong headers. Thank you. 

The problem of "libx11-dev must be installed" persists... Now the compilation reports:
<pre>
Checking for C++ library X11... (cached) no
libx11-dev must be installed!
scons: done reading SConscript files.
scons: Building targets ...
g++ -o xboxdrv -pthread src/main/main.o libxboxdrv.a -ldbus-glib-1 -ldbus-1 -lgobject-2.0 -lgthread-2.0 -lglib-2.0 -lusb-1.0 -ludev
libxboxdrv.a(evdev_helper.o): In function `X11KeysymEnum::process_keymap(_XDisplay*)':
/root/rpmbuild/BUILD/xboxdrv-linux-0.8.5/src/evdev_helper.cpp:73: undefined reference to `XDisplayKeycodes'
/root/rpmbuild/BUILD/xboxdrv-linux-0.8.5/src/evdev_helper.cpp:79: undefined reference to `XGetKeyboardMapping'
/root/rpmbuild/BUILD/xboxdrv-linux-0.8.5/src/evdev_helper.cpp:92: undefined reference to `XKeysymToString'
/root/rpmbuild/BUILD/xboxdrv-linux-0.8.5/src/evdev_helper.cpp:106: undefined reference to `XFree'
libxboxdrv.a(evdev_helper.o): In function `X11KeysymEnum::X11KeysymEnum()':
/root/rpmbuild/BUILD/xboxdrv-linux-0.8.5/src/evdev_helper.cpp:57: undefined reference to `XOpenDisplay'
/root/rpmbuild/BUILD/xboxdrv-linux-0.8.5/src/evdev_helper.cpp:65: undefined reference to `XCloseDisplay'
collect2: error: ld returned 1 exit status
scons: *** [xboxdrv] Error 1
scons: building terminated because of errors.
</pre>

Grumbel, 2015-08-28:
Check config.log that should tell why the x11 check fails.

alejandro-anv, 2015-08-28:
<pre>
file /root/rpmbuild/BUILD/xboxdrv-linux-0.8.5/SConstruct,line 129:
        Configure(confdir = .sconf_temp)
scons: Configure: Checking for C++ library X11... 
scons: Configure: ".sconf_temp/conftest_0.cpp" is up to date.
scons: Configure: The original builder output was:
  |.sconf_temp/conftest_0.cpp <-
  |  |
  |  |
  |  |#include "X11/Xlib.h"
  |  |
  |  |int
  |  |main() {
  |  |
  |  |return 0;
  |  |}
  |  |
  |
g++ -o .sconf_temp/conftest_0.o -c -isystem/usr/include/dbus-1.0 -isystem/usr/lib64/dbus-1.0/include -isystem/usr/include/glib-2.0 -isystem/usr/lib64/glib-2.0/include -pthread -isystem/usr/include/libusb-1.0 -g -O3 -Wall -ansi -pedantic -DPACKAGE_VERSION='"0.8.5"' -Isrc .sconf_temp/conftest_0.cpp
g++: error: .sconf_temp/conftest_0.cpp: No such file or directory
g++: fatal error: no input files
compilation terminated.
scons: Configure: no
</pre>

But:
1) the directory .sconf_temp is empty. g++ says it can't find  .sconf_temp/conftest_0.cpp and it is true.
2) I manually created the conftest_0.cpp and run the g++ compilation manually, and it worked without problem.
3) whith the file manually created, scons worked Ok and compiled the program... Problem solved, but I don't like problems what are solved by themselves....

Thank you for your help.

Grumbel, 2015-11-03:
Could be a permission problem, if .sconf_temp/ was owned by root a user would not be allowed to create files or something like that.

closed 2015-11-03 by Grumbel