xboxdrv/xboxdrv #141: 100% CPU usage when gamepad disconnected at Ubuntu 15.04

NicholasShatokhin, 2015-08-31:
When I disconnect device, xboxdrv become load CPU for 100%. Connecting device back will not help.

Also, if xboxdrv launching set at /etc/rc.local and device don't connected, Ubuntu will never boot.

Fenisu, 2015-10-03:
I am also having this 100% CPU usage issue if I disconnect my gamepad under Arch.

ryukinix, 2017-10-31:
I have the same issue too on Artix Linux (Arch Linux mini-frok systemd-free).

ryukinix, 2017-11-07:
You can solve that with a dirty hack creating a `udev` rule to kill the `xboxdrv` process when you disconnect your controller. This solved for me as I did [here](https://www.github.com/ryukinix/xboxdrv-bin) for my user-case. 

All you need to do that is knowing your VENDORID and PRODUCTID of your hardware, which can be fetched by the `lsusb` on terminal like: 

```
❯ lsusb
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 008 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 007 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 001 Device 005: ID 1e4e:0102 Cubeternet GL-UPC822 UVC WebCam
Bus 001 Device 002: ID 058f:6364 Alcor Micro Corp. AU6477 Card Reader Controller
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 006 Device 002: ID 2516:0011
Bus 006 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 005 Device 002: ID 046d:c246 Logitech, Inc. Gaming Mouse G300
Bus 005 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 004 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 003 Device 002: ID 0810:0001 Personal Communication Systems, Inc. Dual PSX Adaptor
Bus 003 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
```
And create a udev rule to this specific hardware, let's say on the file: `/etc/udev/rules.d/61-xboxdrv-dirty-hack.rules`. To create correct rule only for this event (you disconnecting your controller) you need to write specific some things. In my case, as I found on the `lsusb` output, my VENDORID=0810 and PRODUCTID=0001, so the udev rule will be something like that:

```
ACTION=="remove", SUBSYSTEMS=="usb", KERNEL=="event*", ATTRS{idVendor}=="0810", ATTRS{idProduct} == "001" == SUBSYSTEM=="input",  RUN+="/usr/bin/pkill xboxdrv"
```

The complete example on my repository is in someway more complex because my user-case is just plug the controller, attach a new driver (xboxdrv) running the process as daemon and OpenRC service, remove the controller, stop the service and ensure that the process was really killed (sometimes xboxdrv bug even more (?)). That after the setup I never worried more about the existence of xboxdrv (calling manually, stopping manually etc)

selurvedu, 2017-11-19:
@ryukinix may be related: #239.

ryukinix, 2017-11-19:
@selurvedu nice debug, I'm solving in a similar way, but I did not debug as you did.

selurvedu, 2017-11-22:
@ryukinix I'm using a systemd unit tied to an udev rule, so systemd starts a new instance of xboxdrv each time I plug the controller in (one instance per controller), and stops it with a SIGHUP as soon as I unplug it. It works quite well, and if xboxdrv fails for some reason, getting it back to work is as simple as reattaching the gamepad.

selurvedu, 2017-11-22:
@ryukinix I just saw that your system is systemd-free, and yep, everything I mentioned is possible by using udev only.

ryukinix, 2017-11-22:
> @ryukinix I just saw that your system is systemd-free, and yep, everything I mentioned is possible by using udev only.

Yes, I do pretty the same what you described but for another init system: OpenRC. Seems we have a complete solution for all type of users. Maybe we should describe that on wiki? May be helpful for another people.