xboxdrv/xboxdrv #142: Xboxdrv 0.8.4 RasPi 2 input bugs.

pancakeykakes, 2015-10-11:
I have submitted issues about this on the RetroPie github and forums, but they are all convinced this is an Xboxdrv issue.

I am using my TriPi scripts (github.com/pancakeykakes/TriPi)
In order to have my controller launch RetroPie or Kodi, depending upon which button I have the scripts assigned to.

My issue is, when launching Emulationstation from those scripts (emu.sh specifically), which is bound to RB, I get input bugs where everything the controller does also controls the terminal in the background behind ES.

Here is a video showing the issue.
https://vid.me/Vo3pp

Is this an actual bug or is there something in my script that I should change?

Raspberry Pi 2 model B
OC'd to RPi2 preset from Raspi config
gpu_mem=320
xboxdrv 0.8.4 (the version sudo-apt get provides)
Xboxdrv arguments being used to launch are in the emu.sh script in my TriPi repository.

Grumbel, 2015-10-21:
The video link doesn't work. As for the controlling stuff, the way it generally works with an X server is:

* keyboard and mouse input go to the window that has focus
* joystick input on the other side goes to whoever wants it, it doesn't go through the X server at all, but is directly captured from /dev/input/eventX

An application can grab the joystick event device exclusively (`EVIOCGRAB`), but that's something the application must do, not xboxdrv.

As a further complication: OpenGL and video playback on the Raspberry Pi by passes the X server completely, so OpenGL/video applications tend to either use /dev/input/eventX or the Terminal for input, which means input can go to multiple applications at once.

I am not sure if there is any good easy fix for this. Some stuff I can think of:

* use `openvt` to switch away from X and to a virtual terminal, this should fix mouse issues
* use `xinput` to disable input devices in X, so the game can use them without interference, need to reenable them once the game has quit
* create a dummy window in X that grabs all input until the game has ended
* kill the background app, run the game, restart the background up when game has ended

closed 2015-10-21 by Grumbel

pancakeykakes, 2015-10-21:
I appreciate your response. 
Since the issue only happens when RetroPie is launched via script, regardless of if xboxdrv is involved in the launch, I have decided on my own that it is an issue with the way RetroPie is launched-- whether they want to admit it or not.

The solution that worked for me and the project I am working on is to have RP launch at boot. This is the only time it does not happen. From there, I added a script to launch kodi by selecting an option in the "Ports" section. everything works as expected and my next revision of my project will be incorporating this.