xboxdrv/xboxdrv #144: The 60 seconds delay

johnnygroovy, 2015-10-24:
Hi ! 

When I launch xboxdrv, I have the normal output :

"Your Xbox/Xbox360 controller should now be available as:
  /dev/input/js1
  /dev/input/event22
Press Ctrl-c to quit
[DEBUG] XboxdrvMain::run(): launching thread
[DEBUG] XboxdrvMain::run(): launching main loop"

But my gamepad doesn't work for exactly 60 seconds (I checked).
After that, it works as expected.

I tested xboxdrv with my PS3 controller (wireless and usb), with my Cyborg Saitek (usb)... I have this 60 seconds delay whatever I try...

I have xboxdrv 0.8.5-3 on archlinux.

Grumbel, 2015-10-28:
It's probably hanging inside of `libusb_handle_events()`, which has a 60sec timeout. No idea why. You could try the 0.8.7 release to see if that helps.

johnnygroovy, 2015-11-01:
I tried with the 0.8.7, but I have the exact same problem.
Thanks for the reply anyway ;)

Grumbel, 2015-11-03:
What is the exact command line you use to start xboxdrv?

johnnygroovy, 2015-11-03:
For my PS3 controller I use this : 

xboxdrv --debug -s --evdev /dev/input/$1 -d --type xbox360-wireless --mimic-xpad-wireless --evdev-absmap ABS_X=x1,ABS_Y=y1,ABS_Z=x2,ABS_RZ=y2,ABS_#48=lt,ABS_#49=rt --evdev-keymap BTN_THUMB=tl,BTN_THUMB2=tr,BTN_BASE5=lb,BTN_BASE6=rb,KEY_#302=a,KEY_#301=b,BTN_DEAD=x,KEY_#300=y,BTN_TRIGGER=back,KEY_#720=guide,BTN_TOP=start,BTN_TOP2=du,BTN_PINKIE=dr,BTN_BASE=dd,BTN_BASE2=dl --axismap -Y1=Y1,-Y2=Y2 --deadzone 15%

Grumbel, 2015-11-03:
What happens when you access the `/dev/input/eventX` device directly without xboxdrv? Still a delay?

Is `xboxdrv -L` slow as well?

Another thing to try: Compile with:

$ scons BUILD=development
$ gdb --args xboxdrv ...
[inside gdb]
(gdb) r
[Hit Ctrl-C when it hangs]
(gdb) where

johnnygroovy, 2015-11-03:
No delay when I access /dev/input/eventX without xboxdrv.

But the result of ```xboxdrv -L``` is quite strange :
```
 id | wid | idVendor | idProduct | Name
----+-----+----------+-----------+--------------------------------------

no controller detected
```

Grumbel, 2015-11-03:
The `xboxdrv -L` output is ok, that only lists controllers directly supported, not ones that you have to specify with `--evdev`. Does the `xboxdrv -L` output come instantly or is it slow as well? Have you tried the `gdb` command?

johnnygroovy, 2015-11-03:
Here it is :

```
Your Xbox/Xbox360 controller should now be available as:
  /dev/input/js1
  /dev/input/event22

Press Ctrl-c to quit
[DEBUG] XboxdrvMain::run(): launching thread
[DEBUG] XboxdrvMain::run(): launching main loop
^C
Program received signal SIGINT, Interrupt.
0x00007ffff5f2e18d in poll () from /usr/lib/libc.so.6
(gdb) where
#0  0x00007ffff5f2e18d in poll () from /usr/lib/libc.so.6
#1  0x00007ffff6ff0235 in ?? () from /usr/lib/libusb-1.0.so.0
#2  0x00007ffff6ff11c3 in libusb_handle_events_timeout_completed () from /usr/lib/libusb-1.0.so.0
#3  0x00007ffff6ff12af in libusb_handle_events () from /usr/lib/libusb-1.0.so.0
#4  0x00000000004a978b in USBGSource::on_source (this=<optimized out>) at src/usb_gsource.cpp:177
#5  USBGSource::on_source_wrap (userdata=<optimized out>) at src/usb_gsource.hpp:55
#6  0x00007ffff7249c7a in g_main_context_dispatch () from /usr/lib/libglib-2.0.so.0
#7  0x00007ffff724a020 in ?? () from /usr/lib/libglib-2.0.so.0
#8  0x00007ffff724a342 in g_main_loop_run () from /usr/lib/libglib-2.0.so.0
#9  0x0000000000423566 in XboxdrvMain::run (this=this@entry=0x7fffffffdcb0) at src/xboxdrv_main.cpp:214
#10 0x0000000000408d4d in Xboxdrv::run_main (this=this@entry=0x7fffffffdfdf, opts=...) at src/xboxdrv.cpp:202
#11 0x00000000004081e8 in Xboxdrv::main (this=this@entry=0x7fffffffdfdf, argc=argc@entry=17, argv=argv@entry=0x7fffffffe0d8) at src/xboxdrv.cpp:400
#12 0x0000000000408114 in main (argc=17, argv=0x7fffffffe0d8) at src/main/main.cpp:24
(gdb)
```

johnnygroovy, 2015-11-03:
(And yes the ```xboxdrv -L``` output comes instantly.)

johnnygroovy, 2015-11-18:
Just for you to know, I just tried to compile the 1.8.8 version to check, and I have the same results.

techwiz24, 2015-11-28:
FYI: This does not appear to be controller specific. I encounter this in `0.8.7-1` and in the latest on the `stable` branch with my ASUS Nexus Player gamepad.

Nio01, 2015-12-25:
Same bug here with Logitech Rumblepad 2.
With evtest the gamepad access is ok, with xboxdrv it works after 60 seconds.

buxit, 2016-02-29:
this seems to work for me:
``` C++
gboolean
USBGSource::on_source()
{
  struct timeval to;
  to.tv_sec = 0;
  to.tv_usec = 0;
  libusb_handle_events_timeout_completed(NULL, &to, NULL);
  return TRUE;
}
```

petr666, 2016-03-07:
@buxit Great, thanks for patch, works also for me

chewi, 2016-07-02:
Thanks @buxit, that works here too.

pisto, 2016-11-01:
Why is this closed? I don't see any fix.

chewi, 2016-11-02:
It's not closed, it's #191 that's closed.

buxit, 2016-11-02:
I have just opened PR #214 with the above "fix".

I wonder if this project is still active at all — there have been no commits since nov 2015, even though there are a couple of open PRs.

chewi, 2016-11-02:
It's a shame as this project is fantastic. I am planning to maintain a fork with the patches that I am easily able to verify as a basis for the package in Gentoo Linux. I won't be doing any significant development on it though, too many responsibilities elsewhere.

pisto, 2016-11-02:
@buxit , have you experimented with disabling the usb layer when not needed, or more simply add a flag to explicitly disable the usb layer no matter what (sort of expert option)? Naively I'd say you just need to suppress [this line](https://github.com/xboxdrv/xboxdrv/blob/stable/src/xboxdrv.cpp#L200), but I don't really know the structure of xboxdrv to say.

buxit, 2016-11-02:
i haven not tested this or used xboxdrv at all for some time as i have since replaced the board of the controller with an adafruit trinket pro:
[buxit/Pro_Trinket_USB_Gamepad](https://github.com/buxit/Pro_Trinket_USB_Gamepad)

(the original hardware was bad and inaccurate and i wanted to add buttons and an additional small joystick anyway)

chewi, 2016-12-04:
I've found that xboxdrv occasionally crashes when disconnecting. I've only looked at one core dump so far but it's failing on the other call to `libusb_handle_events` in usb_controller.cpp. This is the only other instance of `libusb_handle_events` in the entire code. Think we need to give this one the same treatment? I can give it a try anyway.

chewi, 2016-12-27:
I've figured out the above issue and it was a little more complicated. I've hopefully fixed it in #220.

robertxgray, 2017-07-06:
Same problem here. Fedora 25 and this gamepad:
https://www.amazon.de/Gamepad-Vibration-Controller-schwarz-Windows/dp/B00BUNOOHQ

Will this ever be fixed? I'll have to continue using AntiMicro for now :(