xboxdrv/xboxdrv #146: XBox One Wireless Adapter Support

Queuecumber, 2015-10-31:
I figured I'd open an issue for this, I have the wireless adapter "for windows 10", I haven't done much debugging with xboxdrv but I figured support for this is one of your goals. I can help debug this and/or provide any information you might need.

marzman, 2015-11-21:
I have a Mac and run windows 10 on a virtual machine -- Is there a way to get the adapter to work with this configuration -- I can connect the adapter successfully to windows 10 but have not been able to get the controller to pair with the adapter -- thanks for any help

steverweber, 2015-12-14:
@marzman here is a mac driver thread to subscribe to.
https://github.com/360Controller/360Controller/issues/156

steverweber, 2015-12-14:
@marzman opps i see you already in there :)

Queuecumber, 2015-12-14:
Lots of good info in those threads including some packet snifs (from https://github.com/360Controller/360Controller/issues/160#issuecomment-159952216)
 
Copied them here for convenience 

https://drive.google.com/file/d/0B381VMfcesrlaG5fTjl3WjZuc2c/view
https://drive.google.com/file/d/0B381VMfcesrlNWZwRkpoTXJBaDA/view
https://drive.google.com/file/d/0B381VMfcesrlRmlKN0tPWGVfdFU/view

mos8580, 2015-12-23:
Hi,

I'm interested in this and own the hardware if any testing/debugging is needed.

toma678, 2016-01-25:
Any news on this? I bought one for cheap, so have the hardware and would be willing to write the code if someone could point me in the right direction?

ryanvade, 2016-09-20:
Is this dead?

WalrusCode, 2016-11-07:
@ryanvade 
This may be late but I believe there is some activity on the xpad git.