xboxdrv/xboxdrv #147: Mad Catz Foghtpad not recognized as a XBox 360 gamepad

terzag, 2015-11-02:
Disclaimer: I use xboxdrv 0.8.5 from the Ubuntu repos. Not sure it's up-to-date so maybe my problem comes from an outdated version.

I have a Mad Catz Fightpad (wired) that is recognized as a generic Xbox controller and doesn't work in some games (the ones that manage only XBox 360 controllers). It works fine when running xboxdrv with the following options:
sudo xboxdrv --device-by-id 1bad:f02e --type xbox360 --detach-kernel-driver --mimic-xpad

Is this behaviour expected or is it a lack of support for this controller by the xboxdrv version I'm using?

In case it's missing support, here is the lsusb output so it can be added in a next update:
Bus 001 Device 023: ID 1bad:f02e Harmonix Music

(If it matters for the description, it's a Mad Catz Fightpad Street Fighter X Tekken edition.)

terzag, 2015-11-02:
For a quick test, I added the ID to xpad_device.cpp and compiled xboxdrv. The controller is now recognized without specifying its ID but it still needs the --mimic-xpad option to be used in games that want a 360 pad.

terzag, 2015-11-04:
Ok, so it's been added but when I did it myself two days ago and compiled xboxdrv, I still had to tweak the options (i.e. specify --mimic-xpad) to use my controller with all games. It is supposed to be the normal behaviour?

Grumbel, 2015-11-04:
Yes, that's normal behavior. You can configure controllers in Steam's Big Picture Mode.

terzag, 2015-11-04:
Oh, ok. So it depens on Steam... :-/

Would it be technically possible to use config files for controllers that xboxdrv would automatically read so that they can be properly configured per basis?

Grumbel, 2015-11-04:
It's not really a Steam thing, it's an SDL2 thing, you can get the same effect by setting the environment variable  SDL_CONTROLLERCONFIG (which is read by SDL2, Unity, UnrealEngine and some more). Steam's Big Picture Mode is just by far the easiest way to do that at the moment, as there isn't another GUI tool to configure that variable. See also:

https://github.com/gabomdq/SDL_GameControllerDB

Also whenever this is a problem, chances are you should just be using xpad and not xboxdrv. xboxdrv is only really useful when you want to customize the input, if you just want something that works xpad is the better choice.

terzag, 2015-11-04:
Ok. In fact, I tried xboxdrv because my Fightpad wasn't recognized by some games when using Xpad. After fiddling with the options and someone pointing me to the --mimic-xpad one, it worked. I'm a bit lost in all of this, I'll check the SDL variable. Someone pointed me to an article about pad configuration through SDL but I assumed it was specific to xboxdrv, so I guess I'm mistaken.

Grumbel, 2015-11-05:
Looks like the situation is a little more complicated then expected. Steam's Big Picture Mode no longer has the configuration menu to set up the controller mapping for some reason.

This variable here should xboxdrv without --mimic-xpad get properly recognized by SDL:

    export SDL_GAMECONTROLLERCONFIG="0000000058626f782047616d65706100,Xbox Gamepad (userspace driver),platform:Linux,a:b0,b:b1,x:b2,y:b3,start:b7,back:b6,guide:b8,dpup:h0.1,dpdown:h0.4,dpleft:h0.8,dpright:h0.2,leftshoulder:b4,rightshoulder:b5,lefttrigger:a5,righttrigger:a4,leftstick:b9,rightstick:b10,leftx:a0,lefty:a1,rightx:a2,righty:a3"

commit 9008740c (2015-11-04): Added Mad Catz Fightpad

Fixes #147
commit 69da6930 (2015-11-04): Added Mad Catz Fightpad

Fixes #147