xboxdrv/xboxdrv #157: Elcom JC-U3613M support

flight16, 2015-11-04:
I just got my XBox 360 gamepad clone working with xboxdrv.  Here is the info: 

> $ lsusb
> Bus 001 Device 005: ID 056e:2004 Elecom Co., Ltd

> osmc@osmc:~$ sudo /usr/bin/xboxdrv --device-by-id 056e:2004 --type xbox360
>
> xboxdrv 0.8.5 - http://pingus.seul.org/~grumbel/xboxdrv/ 
> ...
> Controller:        unknown
> Vendor/Product:    056e:2004
> USB Path:          001:005
> Controller Type:   Xbox360
>
> Your Xbox/Xbox360 controller should now be available as:
>  /dev/input/js0
>  /dev/input/event3

Can this controller and ID be added to xboxdrv?

closed 2015-11-04 by Grumbel

flight16, 2015-11-04:
Thanks for such a quick addition.

I'm sorry, I had a typo in the company name.  It should be "Elecom", not "Elcom."

Grumbel, 2015-11-05:
Fixed.

commit d13a6911 (2015-11-04): Added Elcom JC-U3613M

Fixes #157
commit 691af2f1 (2015-11-04): Added Elcom JC-U3613M

Fixes #157