xboxdrv/xboxdrv #168: Wireless xbox controllers get weird udev device indexes

UglyBob79, 2015-11-29:
Not sure if it is a bug or maybe just some configuration needed, but I have tried to fix this for weeks now so...

Anyway, I'm using 4 x wireless xbox 360 controllers with an original MS receiver in arch linux. When I power up the controllers the player leds light up in the correct order, 1, 2, 3, 4. Then when I use them in retroarch with udev input driver, the controllers gets this weird device index order:

Led 1: device 3
Led 2: device 0
Led 3: device 1
Led 4: device 2

Not sure where the problem is or how I can fix it.

UglyBob79, 2015-11-30:
Could it be that the driver expects them to be indexed from 1 to 4 and subtracts one, but as it is really from 0 to 3 the first one wraps to 3 and the other ones get lowered by one?

theguardian, 2016-03-13:
I have this problem too.

https://groups.google.com/forum/#!topic/xboxdrv/o4lEfw5-A5E

theguardian, 2016-03-13:
This is really goofy, but I came up with a workaround. If you're using "Method 1" as outlined here: https://github.com/retropie/retropie-setup/wiki/Setting-up-the-XBox360-controller

Then instead of going in order from id 0, id 1, id 2, id 3 - you need to change the order in rc.local so that the last is listed first. Thusly:
* id 3
* id 0
* id 1
* id 2

Worked for me. This has been driving me insane for a long time. YMMV

UglyBob79, 2016-03-14:
Ah ok, thanks for sharing! Will test it next time I use my emulators. :)