xboxdrv/xboxdrv #170: All evdev controllers have inverted Y

TiZ-EX1, 2015-12-02:
Hi there. I use xboxdrv to turn all of my controllers into xbox controllers so that they have consistent button layouts across ALL applications, not just SDL2 ones, and so that Unity games don't ignore them.

It works well except that every joystick I've used so far requires the Y axis to be inverted on both sticks for it to work properly. I have an axismap configuration for doing that, but it causes conflicts with any other config I try to add on top that does axismaps, like mapping the dpad to ls.

Because of that and just on general principle, I feel this should be fixed at the source.

My computers are 64-bit, running Xubuntu and Arch. This happens on all of them, on every distro version, and both xboxdrv 0.8.5 and 0.8.8. It's present in older versions too but I can't say exactly how far back. What other information do you need?

piedar, 2015-12-16:
I see the same effect, with two different evdev controllers.  The obvious workaround is

    [axismap]
    -Y1 = Y1
    -Y2 = Y2

TiZ-EX1, 2015-12-17:
That's the workaround I mentioned in my OP. But if I try to add a config on top of that that maps the dpad to LS--like for old Windows games that read the X/Y axes but could just as well be played with dpad--then vertical inputs don't work. We could fix that issue, but that's a non-issue if this one is fixed.

piedar, 2015-12-17:
I suspect it is caused here in [uinput_config.cpp](https://github.com/xboxdrv/xboxdrv/blob/dd9e9b23152a3a5d2d7d27ac169b57e2d4a2326e/src/uinput_config.cpp#L120).  Maybe the developer tested with some evdev controller where inverting the Y values is necessary.  But if most controllers aren't like that, it does not make sense to hard code it.

TiZ-EX1, 2015-12-17:
That looks more to me like Xbox360 controllers--like if you connect them through xboxdrv which uses raw usb access instead of using the xpad driver--have inverted Y axes and thus it's necessary to send them that way so they appear to the system correctly.

@Grumbel, can you comment on this?

Grumbel, 2015-12-17:
The problem is that the events from the evdev controller get squished into the Xbox360Msg struct. It's basically a big ugly hack that doesn't work when the controller doesn't behave very similar to an Xbox360 controller, this invert issue is really just the tip of the ice berg.

The `develop` version handles things a little cleaner and gets rid of a hardcoded Xbox360Msg struct, but that `develop` version is currently stuck in the "I have more interesting things to work on"-hell, so it's not working to great either.

I can't think of any non-ugly quick workaround right now, as the Evdev controller handling doesn't know which axis would need to be inverted, as that depends on how the user maps things.

So for the time being just roll with the workaround:

    [axismap]
    -Y1 = Y1
    -Y2 = Y2

TiZ-EX1, 2015-12-17:
Okay. Thanks for the explanation. :)