xboxdrv/xboxdrv #172: Adding Power A Xbox One Mini Wired Controller [24c6:541a]

illwieckz, 2015-12-18:
Hi, I'm trying to add support for the [Power A Xbox One Mini Wired Controller](http://powera.com/Products/PROD_CTRLS/PLATFORM_XBOXONE/CPFA115320-01).

I made this change:

```
$ git diff
diff --git a/src/xpad_device.cpp b/src/xpad_device.cpp
index 2fc5cf9..bbcd642 100644
--- a/src/xpad_device.cpp
+++ b/src/xpad_device.cpp
@@ -126,6 +126,7 @@ XPadDevice xpad_devices[] = {
   { GAMEPAD_XBOX360,          0x24c6, 0x5d04, "Razer Sabertooth" },
   { GAMEPAD_XBOX360,          0x0e6f, 0x011f, "Rock Candy Gamepad Wired Controller" },
   { GAMEPAD_XBOX360,          0x0e6f, 0x021f, "Rock Candy Gamepad for Xbox 360" },
+  { GAMEPAD_XBOX360,          0x24c6, 0x541a, "Power A Xbox One Mini Wired Controller" },
 
   { GAMEPAD_XBOX_MAT,         0x0738, 0x4540, "Mad Catz Beat Pad" },
   { GAMEPAD_XBOX_MAT,         0x0738, 0x6040, "Mad Catz Beat Pad Pro" },
```

But when I run `xboxdrv` as root I get:

```
# ./xboxdrv
xboxdrv 0.8.8 - http://pingus.seul.org/~grumbel/xboxdrv/ 
Copyright © 2008-2011 Ingo Ruhnke <grumbel@gmail.com> 
Licensed under GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html> 
This program comes with ABSOLUTELY NO WARRANTY. 
This is free software, and you are welcome to redistribute it under certain conditions; see the file COPYING for details. 

Controller:       Power A Xbox One Mini Wired Controller
Vendor/Product:    24c6:541a
USB Path:          001:005
Controller Type:   Xbox360

-- [ ERROR ] ------------------------------------------------------
USBController::USBController(): libusb_open() failed: LIBUSB_ERROR_ACCESS
```

This is the output of `lsusb`:

```
$ lsusb -v -d 24c6:541a

Bus 001 Device 005: ID 24c6:541a  
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass          255 Vendor Specific Class
  bDeviceSubClass       255 Vendor Specific Subclass
  bDeviceProtocol       255 Vendor Specific Protocol
  bMaxPacketSize0        64
  idVendor           0x24c6 
  idProduct          0x541a 
  bcdDevice            1.01
  iManufacturer           1 BDA
  iProduct                2 Xbox ONE mini
  iSerial                 3 00003D9445422EDD
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           32
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass     71 
      bInterfaceProtocol    208 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               4
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               4
Device Status:     0x0002
  (Bus Powered)
  Remote Wakeup Enabled
```

Do you have an idea about what is going wrong?

Grumbel, 2015-12-18:
XboxOne controller are not supported in the stable branch. 

The develop branch does have some XboxOne support, but only wireless it seems (i.e. GAMEPAD_XBOXONE_WIRELESS), no idea if that will work with wired controller.

You might have more luck with the xpad.c kernel driver, I think this one is the latest one:

https://github.com/ValveSoftware/steamos_kernel/blob/brewmaster-4.1/drivers/input/joystick/xpad.c

illwieckz, 2015-12-18:
Oh thanks a lot, my patch that wasn't working against the vanilla kernel source tree is working against this source tree.

closed 2015-12-18 by illwieckz