xboxdrv/xboxdrv #176: Add support for Natec Genesis P44

d4nk4n, 2015-12-25:
(sorry i have no idea how to format this, also this is my first time ever posting somthing on github so i will be grateful for any pointers on what to include).
command to emulate :

xboxdrv --evdev /dev/input/by-id/usb-SZMY-POWER_CO._LTD._EX10_GAMEPAD-event-joystick --evdev-absmap ABS_X=x1,ABS_Y=y1,ABS_Z=x2,ABS_RZ=y2,ABS_HAT0X=dpad_x,ABS_HAT0Y=dpad_y --evdev-keymap BTN_B=a,BTN_C=b,BTN_A=x,BTN_X=y,BTN_Y=lb,BTN_Z=rb,BTN_TL=lt,BTN_TR=rt,BTN_TL2=select,BTN_TR2=start,BTN_SELECT=tl,BTN_START=tr  --axismap -Y1=Y1,-Y2=Y2 --mimic-xpad 
 
After this pad works ok (tested in Dont Starve and Dont Starve Together and Saints Row 4).
And the output of evtest /dev/input/by-id/usb-SZMY-POWER_CO._LTD._EX10_GAMEPAD-event-joystick

Input driver version is 1.0.1
Input device ID: bus 0x3 vendor 0xf0d product 0x9 version 0x110
Input device name: "SZMY-POWER CO.,LTD. EX10 GAMEPAD"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 304 (BTN_SOUTH)
    Event code 305 (BTN_EAST)
    Event code 306 (BTN_C)
    Event code 307 (BTN_NORTH)
    Event code 308 (BTN_WEST)
    Event code 309 (BTN_Z)
    Event code 310 (BTN_TL)
    Event code 311 (BTN_TR)
    Event code 312 (BTN_TL2)
    Event code 313 (BTN_TR2)
    Event code 314 (BTN_SELECT)
    Event code 315 (BTN_START)
    Event code 316 (BTN_MODE)
  Event type 3 (EV_ABS)
    Event code 0 (ABS_X)
      Value    128
      Min        0
      Max      255
      Flat      15
    Event code 1 (ABS_Y)
      Value    128
      Min        0
      Max      255
      Flat      15
    Event code 2 (ABS_Z)
      Value    129
      Min        0
      Max      255
      Flat      15
    Event code 5 (ABS_RZ)
      Value    128
      Min        0
      Max      255
      Flat      15
    Event code 16 (ABS_HAT0X)
      Value      0
      Min       -1
      Max        1
    Event code 17 (ABS_HAT0Y)
      Value      0
      Min       -1
      Max        1
  Event type 4 (EV_MSC)
    Event code 4 (MSC_SCAN)

d4nk4n, 2015-12-25:
Also if someone more knowledgeable would tell me how can I remove/disable the default driver for this pad i would be very grateful , it doesn't appear to use xpad and games detect 2 controllers - second is controlled by xboxdrv and the first one is the default one , if i could remove it ,the pad would work as a perfect xbox 360 pad replacement with xboxdrv. Sorry if this is offtopic.

zerojay, 2016-02-28:
It appears you're using some sort of program that reports itself as a controller here...? What is the actual controller you're using?

d4nk4n, 2016-02-29:
im confused. Im using the actual Natec P44 genesis controller with xboxdrv. it gets detected as a playstation 3 controller on older kernels (3.13) and as /dev/input/by-id/usb-SZMY-POWER_CO._LTD._EX10_GAMEPAD-event-joystick in 3.19 , 4.2 kernels. Please tell me what can i dump here to help you help me.