xboxdrv/xboxdrv #179: 0e6f:0246 controller support (rock candy)

AdamVanOijen, 2016-01-06:
I see in version 0.8.8 you added support for the 0e6f:011f and  0e6f:021f controller, but I seen to have an obscure type of rock candy controller... my one one is listed as 0e6f:0246 via lsusb. please add support :)

zerojay, 2016-02-28:
I found your issue here and decided to add it to my own xboxdrv fork: http://github.com/zerojay/xboxdrv

It would be nice to get the full proper name of the controller as listed from lsusb as well, thanks.

nathankidd, 2017-02-18:
I have this controller and am trying to get it to function.  Using master I've added the `0246` ID (note an `0146` ID is already listed with identical product name).  After `dkms remove` / `dkms install` (per readme) and a modprobe xpad (not in readme, necessary?) I get 

```
[683038.944048] usb 1-5.4.1: new full-speed USB device number 16 using ehci-pci
[683039.038407] usb 1-5.4.1: New USB device found, idVendor=0e6f, idProduct=0246
[683039.038415] usb 1-5.4.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[683039.038420] usb 1-5.4.1: Product: Rock Candy Wired Controller for Xbox One
[683039.038425] usb 1-5.4.1: Manufacturer: Performance Designed Products
[683039.038429] usb 1-5.4.1: SerialNumber: 00008E34DBC60916
[683039.038945] input: Rock Candy Wired Controller for Xbox One as /devices/pci0000:00/0000:00:1d.7/usb1/1-5/1-5.4/1-5.4.1/1-5.4.1:1.0/input/input24
[683039.039975] xpad 1-5.4.1:1.0: xpad_prepare_next_out_packet - found pending output packet 0
[683039.042544] xpad 1-5.4.1:1.0: xpad_prepare_next_out_packet - found pending output packet 0
```

`/dev/input/js0` is created but when I run `jstest /dev/input/js0` no input has any visible effect.  Further, no debug output is listed by dmesg besides the "found pending output packet" lines. There are no LEDs or other indications given by the controller that it is active.

I'm running on Ubuntu 14.04.05 (3.13.0 kernel).

Does anyone have suggestions on how to proceed?  Does "found pending output packet" seem like a blocking issue I should investigate?  Use Wireshark to trace USB?   Thanks for any pointers.