xboxdrv/xboxdrv #180: Trying to use a same evdev to create 2 joypads ... no success

substring, 2016-01-09:
Hi,

I'm trying to map a keyboard as an xbox pad. The reason is to help some people having an ultimarc IPAC2 (usb encoder that maps some inputs to a keyboard device). That IPAC2 can handle 2 players. Inputs from player 1 and player 2 are mapped to keyboard events, on a single keyboard device in linux. 2 players on the same keyboard ... so far so good.

Now, i'm trying to create 2 xbox pads from the same keyboard. I've created and tested 2 config files (one for each player), but when i start xboxdrv to test with `xboxdrv --id=0 --config keyboard.xboxdrv --next-controller --id=1 --config keyboard2.xboxdrv`, xboxdrv creates only 1 joypad, when i expected 2. Is there a way to get 2 pads instead of one ? I know they both grab events from the same /dev/input path, but still, can't I really have 2 pads ? please please pleaaaaaaase

gyates100895, 2016-06-23:
Check out the manual for xboxdrv at http://pingus.seul.org/~grumbel/xboxdrv/xboxdrv.html . In the "Uinput Configuration Options" section, go to the "--ui-buttonmap UIBUTTONSPEC,..." arguement, the fourth section of code seems to be exactly what you need but further reading will help with your specific implemintation.

AzaIndustries, 2018-03-21:
I know this was a long time ago but I found a simple solution that just runs 2 xboxdrvs

`#!/bin/bash

# Bash script for using two (or more) wireless controllers
# ########################################################
#
# Running xboxdrv regularly only supports a single controller, using
# two or more controllers is however possible by launching multiple
# instances of xbdroxv. When launching two instances the build-in
# command execution of xboxdrv no longer can be used, so keeping track
# and killing the xboxdrv processes has to be done manually.

# tell bash to exit the script when something goes wrong
set -e

# kill all existing xboxdrv to prevent duplication
pkill -f xboxdrv &

# remove existing IPAC JS entry
rm -f /dev/input/js*

# launch xboxdrv for the first wireless controller
xboxdrv \
--evdev /dev/input/by-id/usb-Ultimarc_IPAC_2_Ultimarc_IPAC_2_8-event-kbd \
--silent \
--detach-kernel-driver \
--mimic-xpad \
--evdev-no-grab \
--evdev-keymap \
KEY_KP2=tl,KEY_KP3=tr,KEY_KP4=guide,\
KEY_Q=lb,KEY_W=lt,KEY_E=rt,KEY_R=rb,\
KEY_A=a,KEY_S=x,KEY_D=b,KEY_F=y,\
KEY_0=back,\
KEY_P=start,\
KEY_2=dl,\
KEY_1=dr,\
KEY_3=du,\
KEY_4=dd \
--axismap \
dpad_x=x1,-dpad_y=y1 &
XBOXPID1=$! &

sleep 2

# launch xboxdrv for the second wireless controller
xboxdrv \
--evdev /dev/input/by-id/usb-Ultimarc_IPAC_2_Ultimarc_IPAC_2_8-event-kbd \
--silent \
--mimic-xpad \
--evdev-no-grab \
--evdev-keymap \
KEY_KP2=tl,KEY_KP3=tr,KEY_KP4=guide,\
KEY_T=lb,KEY_Y=lt,KEY_U=rt,KEY_I=rb,\
KEY_G=a,KEY_H=x,KEY_J=b,KEY_K=y,\
KEY_9=back,\
KEY_O=start,\
KEY_6=dl,\
KEY_5=dr,\
KEY_7=du,\
KEY_8=dd \
--axismap \
dpad_x=x1,-dpad_y=y1 &
XBOXPID2=$! &
`