xboxdrv/xboxdrv #181: USB device disappearing

IanFin, 2016-02-02:
I'm trying to set up xboxdrv to use with my wired Xbox 360 controller with RetroPie/Emulation Station. I have the following in my rc.local:

xboxdrv --daemon --detach --dbus disabled --detach-kernel-driver --id 0 --led 2 --deadzone 4000 --silent --trigger-as-button

When RetroPie and Emulation Station are starting up one LED on the controller is constant, but then when it finishes loading and asks me to pick an emulator the lights begin to flash and I can't do anything with the controller. When I hit F4 to go back to the terminal, I usually see the following error message repeated multiple times:

[ERROR] XboxdrvDaemon::launch_controller_thread(): USB device disappeared before it could be opened

From the terminal, if I unplug the controller and plug it back in again I get the same message and have to Ctrl+C to get back to the command line.

Given the message, I'm assuming it's a problem with xboxdrv, rather than RetroPie or Emulation Station

Grumbel, 2016-02-03:
Is the xpad kernel driver unloaded and blacklisted? Does 'lsusb' show the device?

IanFin, 2016-02-03:
lsusb is showing the device and I've unloaded and blacklisted xpad as described here

https://29a.ch/2013/2/24/xbox-controller-with-ubuntu-steam-xboxdrv