xboxdrv/xboxdrv #188: src/uinput_options.cpp compile error

sagitter, 2016-03-05:
Hi all,

this error comes out by compiling `xboxdrv-0.8.8` on CentOS6:

```
g++ -o src/uinput_options.o -c -O2 -g -pipe -Wall -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector --param=ssp-buffer-size=4 -m64 -mtune=generic -Wl,-z,relro -fPIC -pie -Wl,-z,now -O2 -g -pipe -Wall -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector --param=ssp-buffer-size=4 -m64 -mtune=generic -Wl,-z,relro -fPIC -pie -Wl,-z,now -isystem/usr/include/dbus-1.0 -isystem/usr/lib64/dbus-1.0/include -isystem/usr/include/glib-2.0 -isystem/usr/lib64/glib-2.0/include -pthread -isystem/usr/include/libusb-1.0 -ansi -pedantic -DPACKAGE_VERSION='"0.8.8"' "-I/usr/include -I/usr/include/X11" -Isrc src/uinput_options.cpp
src/uinput_options.cpp: In member function 'void UInputOptions::mimic_xpad_wireless()':
src/uinput_options.cpp:95: error: 'BTN_TRIGGER_HAPPY3' was not declared in this scope
src/uinput_options.cpp:96: error: 'BTN_TRIGGER_HAPPY4' was not declared in this scope
src/uinput_options.cpp:97: error: 'BTN_TRIGGER_HAPPY1' was not declared in this scope
src/uinput_options.cpp:98: error: 'BTN_TRIGGER_HAPPY2' was not declared in this scope
scons: *** [src/uinput_options.o] Error 1
scons: building terminated because of errors.
error: Bad exit status from /var/tmp/rpm-tmp.xqCZI9 (%build)

```
Please, take a look.
Full log: http://fpaste.org/334494/

GCC-4.4.7
Boost-1.48.0

Grumbel, 2016-03-05:
Your kernel headers are to old. As a quick workaround you can just replace the defines with their values:

    BTN_TRIGGER_HAPPY1		0x2c0
    BTN_TRIGGER_HAPPY2		0x2c1
    BTN_TRIGGER_HAPPY3		0x2c2
    BTN_TRIGGER_HAPPY4		0x2c3

sagitter, 2016-03-05:
Fixed with this patch:

```
--- src/uinput_options.orig.cpp	2015-11-09 11:19:35.000000000 +0100
+++ src/uinput_options.cpp	2016-03-05 18:33:32.308813779 +0100
@@ -92,10 +92,10 @@
   get_axis_map().clear();
   get_btn_map().clear();
 
-  get_btn_map().bind(XBOX_DPAD_UP,    ButtonEvent::create_key(DEVICEID_JOYSTICK, BTN_TRIGGER_HAPPY3));
-  get_btn_map().bind(XBOX_DPAD_DOWN,  ButtonEvent::create_key(DEVICEID_JOYSTICK, BTN_TRIGGER_HAPPY4));
-  get_btn_map().bind(XBOX_DPAD_LEFT,  ButtonEvent::create_key(DEVICEID_JOYSTICK, BTN_TRIGGER_HAPPY1));
-  get_btn_map().bind(XBOX_DPAD_RIGHT, ButtonEvent::create_key(DEVICEID_JOYSTICK, BTN_TRIGGER_HAPPY2));
+  get_btn_map().bind(XBOX_DPAD_UP,    ButtonEvent::create_key(DEVICEID_JOYSTICK, 0x2c2));
+  get_btn_map().bind(XBOX_DPAD_DOWN,  ButtonEvent::create_key(DEVICEID_JOYSTICK, 0x2c3));
+  get_btn_map().bind(XBOX_DPAD_LEFT,  ButtonEvent::create_key(DEVICEID_JOYSTICK, 0x2c0));
+  get_btn_map().bind(XBOX_DPAD_RIGHT, ButtonEvent::create_key(DEVICEID_JOYSTICK, 0x2c1));
 
   get_btn_map().bind(XBOX_BTN_START, ButtonEvent::create_key(DEVICEID_JOYSTICK, BTN_START));
   get_btn_map().bind(XBOX_BTN_GUIDE, ButtonEvent::create_key(DEVICEID_JOYSTICK, BTN_MODE));
```

Thanks a lot.

closed 2016-03-05 by sagitter