xboxdrv/xboxdrv #202: Support for Steam Controllers

johnjelinek, 2016-06-05:
Is there any documentation on mapping multiple steam controllers connected to a single wireless dongle?

chewi, 2016-12-25:
I've just acquired a Stream controller so I'm going to attempt to add native support. It is obviously important to consider this feature when adding such support but I only have one controller. I gather that the Xbox 360 dongle can also do this though and I'm not aware of any support for this in the code. I'll see what that looks like with two controllers.

Grumbel, 2016-12-26:
There is a Open Source driver for the Steam Controller at:

https://github.com/ynsta/steamcontroller

chewi, 2016-12-26:
Just to let you know that I already have the single controller mostly working! :tada: Will polish it off later.

johnjelinek, 2016-12-26:
> working! 🎉 Will polish it off later.

Great news!

chewi, 2016-12-27:
I've pushed up my work so far to my fork under the steam-controller branch.

All the controls are mapped out. The d-pad buttons don't allow you to press up/down together with left/right so I emulated it by checking the analog pad position when the buttons are pressed. 

It should work with wired and multiple wireless controllers in the same manner that multiple 360 controllers are currently supported.

I've been having mixed fortunes with force feedback. I tried to copy what the Python project does and I did get it to work just once but it seems very temperamental. Sometimes the controller won't turn back on until I pop a battery out.