xboxdrv/xboxdrv #218: Xbox360 Wireless Controller couldn't find matching endpoint

Riokei, 2016-12-18:
Alright so I'm trying to set up an xbox360 wireless controller. When I run:
```
sudo xboxdrv --silent --detach-kernel-driver --trigger-as-button --ui-axismap x2=ABS_Z,y2=ABS_RZ --ui-buttonmap A=BTN_B,B=BTN_X,X=BTN_A,TR=BTN_THUMBL,TL=BTN_MODE,GUIDE=BTN_THUMBR
```
I get "No Xbox or Xbox360 controller found" so then I go with the device id lsusb give me:

```
sudo xboxdrv --device-name "Microsoft Corp. Xbox360 Wireless Controller" --device-by-id 045e:028f --type xbox360 --silent --detach-kernel-driver --trigger-as-button --ui-axismap x2=ABS_Z,y2=ABS_RZ --ui-buttonmap A=BTN_B,B=BTN_X,X=BTN_A,TR=BTN_THUMBL,TL=BTN_MODE,GUIDE=BTN_THUMBR
```
I get: "USBController::usb_find_ep(): couldn't find matching endpoint" I checked the notes in #44 but the device isn't showing up as a HID device. I'll attach the output:
[output.txt](https://github.com/xboxdrv/xboxdrv/files/659497/output.txt)

Thanks in advance for any help on this issue.

ThePhisics101, 2017-11-20:
I have the same issue, trying to connect my xbox360 wireless controller with the usb cable
This is what I did: [output.txt](https://github.com/xboxdrv/xboxdrv/files/1487491/output.txt)
Is this a bug or are wired wireless xbox360 controllers not supported?

BatmanofZuhandArrgh, 2023-12-31:
Anyone???