xboxdrv/xboxdrv #225: The output of --help-buttons differs from the input of --evdev-keymap

cpba, 2017-01-05:
`--help-buttons` lists the four dpad directions as:

    DPAD_UP, DPAD_DOWN, DPAD_LEFT, DPAD_RIGHT

However `--evdev-keymap` doesn't recognise these and expects something else instead:

    DU, DD, DL, DR

 It seems like a potential source of confusion. Is there a reason these don't coincide?

anohren, 2017-03-18:
I just raised the same issue...in the wrong repo apparently.

The allowed dpad names seem to be "dd" or "down" etc.:

```
  else if (str == "du" || str == "up")
    return XBOX_DPAD_UP;
  else if (str == "dd" || str == "down")
    return XBOX_DPAD_DOWN;
  else if (str == "dl" || str == "left")
    return XBOX_DPAD_LEFT;
  else if (str == "dr" || str == "right")
    return XBOX_DPAD_RIGHT;

  else
raise_exception(std::runtime_error, "couldn't convert string \"" + str + "\" to XboxButton");
```

but `--help-button` prints

```
XboxButton:
  Start, Guide, Back, A, B, X, Y, LB, RB, LT, RT, TL, TR, DPAD_UP, DPAD_DOWN, 
  DPAD_LEFT, DPAD_RIGHT 
```

Looks like the help is just outdated

cpba, 2017-03-19:
So you're saying that the proper solution would be to update the help?

anohren, 2017-03-19:
I don't know what's a proper solution. I just noticed it when I tried to make it read my config file.

From the snippet you posted the problem is apparent. These two functions don't match each-other:

```
XboxButton string2btn(const std::string& str_)
{
  std::string str = to_lower(str_);

  if (str == "start")
    return XBOX_BTN_START;
  else if (str == "guide" || str == "ps")
    return XBOX_BTN_GUIDE;
  else if (str == "back" || str == "select")
    return XBOX_BTN_BACK;

  else if (str == "a" || str == "1" || str == "green" || str == "cross")
    return XBOX_BTN_A;
  else if (str == "b" || str == "2" || str == "red" || str == "circle")
    return XBOX_BTN_B;
  else if (str == "x" || str == "3" || str == "blue" || str == "square")
    return XBOX_BTN_X;
  else if (str == "y" || str == "4" || str == "yellow" || str == "triangle")
    return XBOX_BTN_Y;

  else if (str == "lb" || str == "5" || str == "orange" || str == "white" || str == "l1")
    return XBOX_BTN_LB;
  else if (str == "rb" || str == "6" || str == "black" || str == "r1")
    return XBOX_BTN_RB;

  else if (str == "lt" || str == "7" || str == "l2")
    return XBOX_BTN_LT;
  else if (str == "rt" || str == "8" || str == "r2")
    return XBOX_BTN_RT;

  else if (str == "tl" || str == "l3")
    return XBOX_BTN_THUMB_L;
  else if (str == "tr" || str == "r3")
    return XBOX_BTN_THUMB_R;

  else if (str == "du" || str == "up")
    return XBOX_DPAD_UP;
  else if (str == "dd" || str == "down")
    return XBOX_DPAD_DOWN;
  else if (str == "dl" || str == "left")
    return XBOX_DPAD_LEFT;
  else if (str == "dr" || str == "right")
    return XBOX_DPAD_RIGHT;

  else
    raise_exception(std::runtime_error, "couldn't convert string \"" + str + "\" to XboxButton");
}

std::string btn2string(XboxButton btn)
{
  switch (btn)
  {
    case XBOX_BTN_MAX:
    case XBOX_BTN_UNKNOWN: return "unknown";

    case XBOX_BTN_START: return "Start";
    case XBOX_BTN_GUIDE: return "Guide";
    case XBOX_BTN_BACK: return "Back";

    case XBOX_BTN_A: return "A";
    case XBOX_BTN_B: return "B";
    case XBOX_BTN_X: return "X";
    case XBOX_BTN_Y: return "Y";

    case XBOX_BTN_LB: return "LB";
    case XBOX_BTN_RB: return "RB";

    case XBOX_BTN_LT: return "LT";
    case XBOX_BTN_RT: return "RT";

    case XBOX_BTN_THUMB_L: return "TL";
    case XBOX_BTN_THUMB_R: return "TR";

    case XBOX_DPAD_UP:    return "DPAD_UP";
    case XBOX_DPAD_DOWN:  return "DPAD_DOWN";
    case XBOX_DPAD_LEFT:  return "DPAD_LEFT";
    case XBOX_DPAD_RIGHT: return "DPAD_RIGHT";
  }
  return "unknown";
}
```

To fix it I suppose I would've defined one set of "official" string representations and one set of aliases

cpba, 2017-03-19:
Serves me well for not checking what the function actually did, I've submitted a pull request adding the missing inputs to string2btn in order to make them match.