xboxdrv/xboxdrv #233: xboxdrv does not generate mouse events

hramrach, 2017-06-25:
xboxdrv --mouse --evdev /dev/input/event9 --silent &

xboxdrv 0.8.8 - http://pingus.seul.org/~grumbel/xboxdrv/ 
Copyright © 2008-2011 Ingo Ruhnke <grumbel@gmail.com> 
Licensed under GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html> 
This program comes with ABSOLUTELY NO WARRANTY. 
This is free software, and you are welcome to redistribute it under certain conditions; see the file COPYING for details. 

Your Xbox/Xbox360 controller should now be available as:
  /dev/input/js2
  /dev/input/event14

evtest
Select the device event number [0-15]: 15
Input driver version is 1.0.1
Input device ID: bus 0x0 vendor 0x0 product 0x0 version 0x0
Input device name: "Xbox Gamepad (userspace driver) - Mouse Emulation"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 272 (BTN_LEFT)
    Event code 273 (BTN_RIGHT)
    Event code 274 (BTN_MIDDLE)
  Event type 2 (EV_REL)
    Event code 0 (REL_X)
    Event code 1 (REL_Y)
    Event code 6 (REL_HWHEEL)
    Event code 8 (REL_WHEEL)

no events seen pressing every button on the gamepad
 
evtest /dev/input/event9
Input driver version is 1.0.1
Input device ID: bus 0x3 vendor 0x46d product 0xc21d version 0x4014
Input device name: "Logitech Gamepad F310"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 304 (BTN_SOUTH)
    Event code 305 (BTN_EAST)
    Event code 307 (BTN_NORTH)
    Event code 308 (BTN_WEST)
    Event code 310 (BTN_TL)
    Event code 311 (BTN_TR)
    Event code 314 (BTN_SELECT)
    Event code 315 (BTN_START)
    Event code 316 (BTN_MODE)
    Event code 317 (BTN_THUMBL)
    Event code 318 (BTN_THUMBR)
  Event type 3 (EV_ABS)
    Event code 0 (ABS_X)
      Value    128
      Min   -32768
      Max    32767
      Fuzz      16
      Flat     128
    Event code 1 (ABS_Y)
      Value   -129
      Min   -32768
      Max    32767
      Fuzz      16
      Flat     128
    Event code 2 (ABS_Z)
      Value      0
      Min        0
      Max      255
    Event code 3 (ABS_RX)
      Value    128
      Min   -32768
      Max    32767
      Fuzz      16
      Flat     128
    Event code 4 (ABS_RY)
      Value   -129
      Min   -32768
      Max    32767
      Fuzz      16
      Flat     128
    Event code 5 (ABS_RZ)
      Value      0
      Min        0
      Max      255
    Event code 16 (ABS_HAT0X)
      Value      0
      Min       -1
      Max        1
    Event code 17 (ABS_HAT0Y)
      Value      0
      Min       -1
      Max        1
  Event type 21 (EV_FF)
    Event code 80 (FF_RUMBLE)
    Event code 81 (FF_PERIODIC)
    Event code 88 (FF_SQUARE)
    Event code 89 (FF_TRIANGLE)
    Event code 90 (FF_SINE)
    Event code 96 (FF_GAIN)

produces some events

Grumbel, 2017-06-27:
When using `--evdev` you have to supply `--evdev-absmap` and `--evdev-keymap` options as well, otherwise xboxdrv won't know how to interpret the events from `/dev/input/event9`.

hramrach, 2017-06-28:
First, there was no complaint from xboxdrv saying I am missing a mapping.

Second, blank mapping is not useful default. I can just do without running xboxdrv in that case.

There is --ui-clear which says removes previous mappings so I assumed there are some default mappings, are there not?

I would expect that the default mapping for a xbox controller would give you a mapping that results in same button assignment as on a xbox-lookalike generic usbhid gamepad, and default evdev mapping is an identity with extra events removed because xboxdrv can only handle what xbox controller has.

Grumbel, 2017-06-28:
> Second, blank mapping is not useful default. 

It's the easy default, because xboxdrv has no idea what events your controller will be sending or how they map to Xbox events. It would be possible to add some automatic guessing or reuse the SDL_GAMECONTROLLER database, but somebody would have to write that code. The whole `--evdev` stuff in xboxdrv is rather underdeveloped.

> There is --ui-clear which says removes previous mappings so I assumed there are some default mappings, are there not?

`--ui-clear` is for how Xbox events are mapped back to uinput/evdev events and for that there are defaults. What you however are looking for is how evdev events are mapped to Xbox events and that's what the `--evdev-*` options are for, not the `--ui-*` ones.

> I would expect that the default mapping for a xbox controller would give you a mapping that results in same button assignment as on a xbox-lookalike generic usbhid gamepad

Anyway, you can forget most of what I have written above, as that only applies to generic input devices, the Logitech 310 is however an Xinput compatible controller, so you don't need to mess around with `--evdev`, it should work directly via:

     sudo xboxdrv --mouse --detach-kernel-driver

hramrach, 2017-06-28:
Using inverse of the default --ui-* for --evdev-* should give pretty much working setup for most generic xbox and ps look-alike gamepads unless the firmware writers were really inventive with non-standard button assignments. All of them should have a dpad and a pair mini joysticks, the colurful buttons, start ans select buttons, and some front buttons. I would expect most of these map to the same events on most gamepads.