xboxdrv/xboxdrv #239: Hangs after getting the controller disconnected, fails to shut down gracefully

selurvedu, 2017-10-11:
I've encountered a strange bug with xboxdrv 0.8.8 which is, unfortunately, not always reproducible. After unplugging the controller, the `xboxdrv` process does not terminate gracefully when pressing Control-C or sending a SIGINT signal.

Steps to reproduce:

1. Plug a gamepad.
2. `/usr/bin/xboxdrv --dbus disabled --silent --config /etc/xboxdrv/myconfig --evdev /dev/input/event7 --debug`
3. Unplug the gamepag.
4. Press Control-C or execute `killall -SIGINT xboxdrv`.
5. It may shut down properly a few times in a row, but eventually you will see it hang. Just repeat steps 1-4 several times. In my case the longest streak of normal behaviour was ~7 attempts.

The output is not helpful:

```
Press Ctrl-C to quit
[DEBUG] XboxdrvMain::run(): launching thread
[DEBUG] XboxdrvMain::run(): launching main loop
^C[INFO]  XboxdrvMain::shutdown(): shutdown requested
^C[INFO]  XboxdrvMain::shutdown(): shutdown requested
^C^C^C
```

Running it under `strace` was painful, but I've managed to get some info:
```
poll([{fd=6, events=POLLIN}, {fd=8, events=POLLIN}, {fd=10, events=POLLIN}, {fd=11, events=POLLIN}, {fd=12, events=POLLIN}, {fd=13, events=POLLIN}], 6, 2) = 2 ([{fd=11, revents=POLLIN}, {fd=12, revents=POLLERR|POLLHUP}])
read(11, "\2\0\0\0\0\0\0\0", 16)        = 8
write(11, "\1\0\0\0\0\0\0\0", 8)        = 8
read(12, 0x7ffccf2b9f90, 3072)          = -1 ENODEV (No such device)
write(11, "\1\0\0\0\0\0\0\0", 8)        = 8
poll([{fd=6, events=POLLIN}, {fd=8, events=POLLIN}, {fd=10, events=POLLIN}, {fd=11, events=POLLIN}, {fd=12, events=POLLIN}, {fd=13, events=POLLIN}], 6, 2) = 2 ([{fd=11, revents=POLLIN}, {fd=12, revents=POLLERR|POLLHUP}])
read(11, "\2\0\0\0\0\0\0\0", 16)        = 8
write(11, "\1\0\0\0\0\0\0\0", 8)        = 8
read(12, 0x7ffccf2b9f90, 3072)          = -1 ENODEV (No such device)
write(11, "\1\0\0\0\0\0\0\0", 8)        = 8
poll([{fd=6, events=POLLIN}, {fd=8, events=POLLIN}, {fd=10, events=POLLIN}, {fd=11, events=POLLIN}, {fd=12, events=POLLIN}, {fd=13, events=POLLIN}], 6, 2) = 2 ([{fd=11, revents=POLLIN}, {fd=12, revents=POLLERR|POLLHUP}])
read(11, "\2\0\0\0\0\0\0\0", 16)        = 8
write(11, "\1\0\0\0\0\0\0\0", 8)        = 8
read(12, 0x7ffccf2b9f90, 3072)          = -1 ENODEV (No such device)
write(11, "\1\0\0\0\0\0\0\0", 8)        = 8
poll([{fd=6, events=POLLIN}, {fd=8, events=POLLIN}, {fd=10, events=POLLIN}, {fd=11, events=POLLIN}, {fd=12, events=POLLIN}, {fd=13, events=POLLIN}], 6, 2) = 2 ([{fd=11, revents=POLLIN}, {fd=12, revents=POLLERR|POLLHUP}])
read(11, "\2\0\0\0\0\0\0\0", 16)        = 8
write(11, "\1\0\0\0\0\0\0\0", 8)        = 8
read(12, 0x7ffccf2b9f90, 3072)          = -1 ENODEV (No such device)
write(11, "\1\0\0\0\0\0\0\0", 8)        = 8
```
(the part above repeats over and over, then I send a SIGINT)
```
--- SIGINT {si_signo=SIGINT, si_code=SI_USER, si_pid=24118, si_uid=0} ---
write(1, "[INFO]  XboxdrvMain::shutdown():"..., 52) = 52
nanosleep({0, 10000000}, 0x7ffccf2ba3a0) = 0
futex(0x1894910, FUTEX_WAIT_PRIVATE, 2, NULL) = ? ERESTARTSYS (To be restarted if SA_RESTART is set)
--- SIGWINCH {si_signo=SIGWINCH, si_code=SI_KERNEL} ---
futex(0x1894910, FUTEX_WAIT_PRIVATE, 2, NULL) = ? ERESTARTSYS (To be restarted if SA_RESTART is set)
--- SIGWINCH {si_signo=SIGWINCH, si_code=SI_KERNEL} ---
futex(0x1894910, FUTEX_WAIT_PRIVATE, 2, NULL) = ? ERESTARTSYS (To be restarted if SA_RESTART is set)
--- SIGWINCH {si_signo=SIGWINCH, si_code=SI_KERNEL} ---
futex(0x1894910, FUTEX_WAIT_PRIVATE, 2, NULL) = ? ERESTARTSYS (To be restarted if SA_RESTART is set)
--- SIGWINCH {si_signo=SIGWINCH, si_code=SI_KERNEL} ---
futex(0x1894910, FUTEX_WAIT_PRIVATE, 2, NULL) = ? ERESTARTSYS (To be restarted if SA_RESTART is set)
--- SIGWINCH {si_signo=SIGWINCH, si_code=SI_KERNEL} ---
futex(0x1894910, FUTEX_WAIT_PRIVATE, 2, NULL
```

Currently I'm using SIGHUP as a workaround. SIGKILL works as well.

ryukinix, 2017-11-19:
Well... SIGKILL is the most powerful signal, this should works always. I do the same. Funny that even SIGINT don't works. You tested with SIGTERM? Actually I expect that the driver shutdown alone without even SIGINT or whatever signal after a disconnection.

selurvedu, 2017-11-22:
> SIGTERM

I don't remember for sure whether it works or not because I chose to stick to SIGHUP. My guess is "no", because otherwise there would be no need to try SIGHUP. I _can_ check that, but it won't change anything — why not just use SIGHUP?

> Actually I expect that the driver shutdown alone without even SIGINT or whatever signal after a disconnection.

Me too. Even though this is only acceptable if xboxdrv is not running in daemon mode.

ryukinix, 2017-11-22:
>  I can check that, but it won't change anything — why not just use SIGHUP?

If SIGHUP works, just use it. SIGHUP and SIGTERM has diferent meaning but sometimes has the same behavior. But sometimes SIGHUP can reload something, which SIGTERM means that the application should kill yourself. SIGKILL don't ask for nothing, just remove the process from the queue of process scheduler of the system. You can read a little more as start if you have interest on it: https://en.wikipedia.org/wiki/Signal_(IPC)#POSIX_signals

But if I was you, I would keep my SIGHUP solution just to forget about this pain... I was expecting just a plug-in-play solution when I came from this, but I passed some days working with it to get a broad and great solution for this set of problems.

Would be nice if this was fixed in the future for other people don't stuck with that.

closed 2026-08-26 by Grumbel