xboxdrv/xboxdrv #242: Controller detected but no signals

Zamundaaa, 2017-10-29:
I'm using Kubuntu 17.10. The controller is detected but no events are generated
This is the output of sudo xboxdrv:
[fail.txt](https://github.com/xboxdrv/xboxdrv/files/1425103/fail.txt)
On Kubuntu 17.04 xpad worked correctly, and still does (tested on another machine). After the upgrade it seemingly failing to read values of my controller so I switched to xboxdrv, which as you see didn't solve the issue. I kind of don't know where else to report this problem

Grumbel, 2017-10-29:
Try running as root:

      sudo xboxdrv

Grumbel, 2017-10-29:
First guess would be that you aren't running as root, but you said you already are.

Second guess would be something wrong with the USB endpoints. Post the output of:

    lsusb -v -d 045e:028e

Output of `dmesg` when using the xpad driver could be interesting as well.

Zamundaaa, 2017-11-03:
I did both and put it in lsusb.txt
[lsusb.txt](https://github.com/xboxdrv/xboxdrv/files/1441213/lsusb.txt)

Zamundaaa, 2017-12-08:
As I've installed libdbus-glib-1-dev and libusb-1.0-0-dev now the error message LIBUSB_TRANSFER_ERROR doesn't appear anymore, but there's still no events being generated

petr666, 2017-12-13:
Maybe it is this problem https://github.com/xboxdrv/xboxdrv/issues/144 ?

Zamundaaa, 2017-12-13:
@petr666 Well I don't think so because I don't get any events whatsoever
But I'll look into it tomorrow. Maybe I'll find something

Grumbel, 2017-12-14:
`xboxdrv --chatpad` would be another thing worth to try.

Zamundaaa, 2017-12-14:
The error message LIBUSB_TRANSFER_ERROR somehow appears again
xboxdrv --chatpad didn't work
[chatpad output.txt](https://github.com/xboxdrv/xboxdrv/files/1559494/chatpad.output.txt)

Zamundaaa, 2017-12-18:
I randomly searched through some forums trying to understand this issue better and when I stumbled on this website: https://forums.dolphin-emu.org/Thread-controller-not-working-ubuntu-17-10-dolphin-5-0-07ab81e-dirty
I just tried what stood there:
sudo apt install libevdev-dev libudev-dev
it seems this missing packages was all what was wrong. Now everything works as expected again

Zamundaaa, 2017-12-18:
Well it seems as though this didn't even survive one reboot. Some things I noticed:
The controllers did work but not that reliable (the reason why I rebooted) and
The amount of buttons displayed by the preference window in KDE plasma is wrong. The controller has 15 buttons and those were displayed when it worked, but now (as before) it shows 11

Zamundaaa, 2017-12-27:
Another thing I found that was weird: when using $ usb-devices whilst using xpad it shows xpad but whilst running xboxdrv it shows me that it's using the usbfs driver for this controller, which isn't even in the linux kernel since ages.

T:  Bus=01 Lev=01 Prnt=01 Port=11 Cnt=01 Dev#= 14 Spd=12  MxCh= 0
D:  Ver= 2.00 Cls=ff(vend.) Sub=ff Prot=ff MxPS=64 #Cfgs=  1
P:  Vendor=045e ProdID=028e Rev=01.10
S:  Manufacturer=Xiaoji
S:  Product=XBOX 360 For Windows
C:  #Ifs= 1 Cfg#= 1 Atr=a0 MxPwr=500mA
I:  If#= 0 Alt= 0 #EPs= 2 Cls=ff(vend.) Sub=5d Prot=01 Driver=usbfs

Zamundaaa, 2017-12-29:
I now have found a pattern. Every time I shut down the PC, turn the power off and wait until the capacitors are all empty (lights are off) and then power it on again with the controller connected, the controller seems to switch into another mode and everything is working. There's four status LEDs on it and if I do so there's two instead of just one enabled. This persists as long as the controller is plugged in, it doesn't matter if I suspend or restart my system