xboxdrv/xboxdrv #255: Error building xboxdrv

gansopro, 2022-07-31:
Hello friends! I've already installed all the necessary requirements, but when I open the source code folder in the terminal and type the scons command, this appears:

scons: Reading SConscript files ...
  File "/home/ganso/Downloads/xboxdrv-linux-0.8.8/SConstruct", line 34

    print target

    ^^^^^^^^^^^^

SyntaxError: Missing parentheses in call to 'print'. Did you mean print(...)?

What do I do now?

tastytea, 2022-07-31:
The file is written in python 2 (which is deprecated and unmaintained) while scons uses python 3 (the current version). 

It looks like the python code was ported to python 3 in the [develop branch](https://github.com/xboxdrv/xboxdrv/tree/develop), so you could try building that.

Another option is to use the the Gentoo patches with the stable branch, you can download them from <https://cgit.gentoo.org/repo/gentoo.git/tree/games-util/xboxdrv/files>.

To apply them, run these commands in the source code folder:

```
wget https://cgit.gentoo.org/repo/gentoo.git/plain/games-util/xboxdrv/files/xboxdrv-0.8.8-Update-SConstruct-to-python3.patch \
     https://cgit.gentoo.org/repo/gentoo.git/plain/games-util/xboxdrv/files/xboxdrv-0.8.8-Updating-python-code-to-python3.patch \
     https://cgit.gentoo.org/repo/gentoo.git/plain/games-util/xboxdrv/files/xboxdrv-0.8.8-some-boost-fix.patch
patch -p1 < xboxdrv-0.8.8-Update-SConstruct-to-python3.patch
patch -p1 < xboxdrv-0.8.8-Updating-python-code-to-python3.patch
patch -p1 < xboxdrv-0.8.8-some-boost-fix.patch
```

gansopro, 2022-07-31:
> To apply them, run these commands in the source code folder:

I typed these three:

wget https://cgit.gentoo.org/repo/gentoo.git/plain/games-util/xboxdrv/files/xboxdrv-0.8.8-Update-SConstruct-to-python3.patch

wget https://cgit.gentoo.org/repo/gentoo.git/plain/games-util/xboxdrv/files/xboxdrv-0.8.8-Updating-python-code-to-python3.patch

wget https://cgit.gentoo.org/repo/gentoo.git/plain/games-util/xboxdrv/files/xboxdrv-0.8.8-some-boost-fix.patch

But in this fourth:

patch -p1 < xboxdrv-0.8.8-Update-SConstruct-to-python3.patch

This appears:

can't find file to patch at input line 14
Perhaps you used the wrong -p or --strip option?
The text leading up to this was:
--------------------------
|From 39a334fbc0482626455f417e97308e52aa8746a7 Mon Sep 17 00:00:00 2001
|From: Ingo Ruhnke <grumbel@gmail.com>
|Date: Sun, 24 Nov 2019 18:16:16 +0100
|Subject: [PATCH 1/3] Update SConstruct to python3
|
|---
| SConstruct | 20 ++++++++++----------
| 1 file changed, 10 insertions(+), 10 deletions(-)
|
|diff --git SConstruct SConstruct
|index 4cd7970..54fa11f 100644
|--- a/SConstruct
|+++ b/SConstruct
--------------------------
File to patch: 

And now, what I do?

gansopro, 2022-07-31:
There, friend, gone. It's just that I didn't have the source code folder open in the terminal.

Okay, but now, should I try to build the software again?

tastytea, 2022-07-31:
Yes, after you patched it, the build instructions from the readme should work.

gansopro, 2022-08-01:
Done, friend, built. Generated xboxdrv file (no extension). What do I do with him? I thought it was going to generate a .deb file, but it didn't.

gansopro, 2022-08-01:
You don't have to answer me, friend. I saw in the readme that now just type "make install" (I typed "sudo make install" because it gave permission denied), and that's what I did and it worked.

Thank you for answering! If I have another question, I'll post another issue, ok.

closed 2022-08-01 by gansopro