xboxdrv/xboxdrv #262: No RSB support?

TRPB, 2024-07-30:
I have a game which only supports Xbox controllers and doesn't allow remapping the controls.

It said "Press RS for [action]" and I couldn't figure out what button that was on a controller.

According to https://gist.github.com/palmerj/586375bcc5bc83ccdaf00c6f5f863e86 it's pressing down on the right stick. There doesn't appear to be a way to map that in xboxdrv: 

> XboxAxis:
  X1, Y1, X2, Y2, LT, RT, DPAD_X, DPAD_Y, TRIGGER, BTN_A, BTN_B, BTN_X, BTN_Y, Black, White
>
> XboxButton:
  Start, Guide, Back, A, B, X, Y, LB, RB, LT, RT, TL, TR, DPAD_UP, DPAD_DOWN, DPAD_LEFT, DPAD_RIGHT

which if any of these is RSB on the linked graphic and is there any way to map it?

Grumbel, 2024-07-30:
BTN_THUMBL/BTN_THUMBR in evdev or TL & TR in xboxdrv, L3/R3 might work too.

TRPB, 2024-08-03:
ah thanks, yes these are TL and TR in xboxdrv

closed 2024-08-03 by TRPB